mlo821/OctoBuddy #7: Feature Request - user defined system commands

Bootscreen, 2021-01-20:
Hey, is it possible to get an Option to Set userdefined System commands e.g. to restart the System or the wifi when i press a Button?

DoWhileGeek, 2021-01-20:
I second the user defined actions. Even if its just within octoprints scope. I would love a hardware button to just start the selected print. Would really help with printing an object, clearing the bed, and printing the same one with minimal fuss.

joergp, 2021-01-24:
Yes, please. The two most important actions for me would be: stop print in octopi and print selected job. My printer is in the basement and I'm always running up and down to select those commands (Octopod is another option, but just pressing a button would be perfect). Other important ones: preheat selected job, ABL and perhaps change filament.