mlo821/OctoBuddy #13: OctoBuddy- keystroke executes multiple commands

FindusD, 2021-05-22:
Hello,
I would like to use OctuBuddy and have printed my case with the buttons.
But now I have problems:

- If I connect just one button to my Raspi, the corresponding function works fine, no matter which one.

- when I connect more than 2 or 3 switches (or all) to the raspi, i have the problem: Every time I press a switch, not only the corresponding function is executed, but always one two others as well.
 
To be more precise: it always executes the functions of the two neighbouring pins as well.
Examples with the default setting:

- I press Z+ (pin 31). Then X-(pin 29), Z+(pin 31), Z-(pin 33) are executed one after the other.

- Or I press Y- (pin 21). Then Y+(pin 19), Y-(pin 21) and X+ (pin 23) are executed.

**Always in ascending pin order.**

I have tested it with 2 Raspberries: with a Raspberry 2 B+ and with a Raspi 3 B.
And I tried it with a circuit on the fly on a breadboard and with my finished OctoBuddy case, with the same result.

I have connected all switches following the description om Factory Restart.
The difference:
My common pin is pin 17 (3.3V), the other pins are 15, 19, 21 etc. as defined in the default settings in the plugin. 
And: There is no short-circuit anywhere!

I have also changed the pin definition in the plugin and connected the switches accordingly, so that there is always at least one unused pin in between, but with the same result.

What could be the reason for this?
Do I need any special setting in the config.txt of my Raspi-SD-card?

Thanks in advance

5ft24dave, 2021-05-23:
Sounds like there is a short in the wiring between the GPIO side of the switches

FindusD, 2021-05-23:
No, there is no short.
I have checked it several times. And it's the same effect when I make a test wiring only with buttons on a breadboard.